psykokwak-com/everblu-meters-esp8266 #2: Questions et Suggestions

xekil, 2022-06-17:
Bonjour et merci une nouvelle fois pour ce projet :)

J'aurais quelques questions et suggestions concernant le fonctionnement :

Avec la dernière version (OTA), j'ai remarqué que l'interrogation du compteur s'effectue à 12:00, et quel ne s'effectue plus lors de la mise en route de l'ESP ou du flash, normal ?

**Lors du test de fréquence, j'ai obtenu 433.861572, dois-je renseigner :**

- 433.8615 ou 433.8615**7** ou 433.8615**72** ?

  Avec 433.8615**7** ça semble fonctionner correctement.

**Est-il possible de rajouter des capteurs MQTT pour Home Assistant :**

- La possibilité d'effectuer une relève manuellement ([peut être via MQTT Button](https://www.home-assistant.io/integrations/button.mqtt/))
- Puissance du signal Wi-Fi ([MQTT Sensor](https://www.home-assistant.io/integrations/sensor.mqtt/))
- État de la connexion ([MQTT Binary Sensor](https://www.home-assistant.io/integrations/binary_sensor.mqtt/))
- Disponibilité de l'ESP (Capteur Uptime)
- Reboot de l'ESP (Switch ou Bouton)

Enfin, avoir un accès au Debug en WIFI serait vraiment bien, j'ai trouvé des projets intéressants et j'ai essayé de les fusionner mais sans succès...

**Terminal via client Telnet**

- https://forum.arduino.cc/t/serveur-telnet-pour-esp8266/656597
- https://www.fais-le-toi-meme.fr/fr/electronique/tutoriel/esp8266-arduinoota-mise-a-jour-logiciel-esp8266-wifi
- https://github.com/JoaoLopesF/RemoteDebug

**Terminal Http**

- https://forum.arduino.cc/t/terminal-wifi-pour-esp/640110
- https://forum.arduino.cc/t/terminal-wifi-pour-esp8266-esp32/643876

PS : Avec ce projet c'était la première fois que j'utilisait Arduino IDE mais je peux aider pour des tests :)
PS2 : J'écris en français car il semble que nous soyons majoritairement des français à utiliser ce projet.

Merci

GSzabados, 2022-06-17:
> PS2 : J'écris en français car il semble que nous soyons majoritairement des français à utiliser ce projet.

I actually do not speak French and use it in Poland, because my city installed Itron AnyQuest Cyble meters...

GSzabados, 2022-06-17:
> * Avec 433.8615**7** ça semble fonctionner correctement.

The meter has a very large tolerance for the weak up signal. (maybe 25kHz) The code does the search with 0.5kHz steps, the original code used a 2kHz step for same purpose.

psykokwak-com, 2022-06-17:
Hi all,
I forked this project to make it working on an ESP8266/ESP32 because this is what I mainly use on my projects.
But this code is for test purposes and I have no goal to make a full featured project.
I'm not sure to improve it unless someone send pull requests.
So, to answer to you questions :
- the latest commit I pushed fixes the polling every day at 10.00am (UTC). You are free to change this according to your needs.
- As @GSzabados said, the meter has à very large tolerence on the frequency. The best solution should be to scan the frequencies at a higher step (I recognize that I was a little too precise) to determine the low frequency and the high frequency then to take the frequency in the middle.
- You can manually trigger an update by publishing (any payload you want) on "everblu/cyble/trigger" topic. So create an entity on HA to handle this.
- I did not plan to implement other functions but if it were to be done, I would rather try to do it on an already existing firmware such as tasmota to take advantage of all the advanced functions of the firmware.

GSzabados, 2022-06-17:
> * determine the low frequency and the high frequency then to take the frequency in the middle.

I did run my device with either 2 or 2.5kHz steps, before I managed to get it right for my meter the wait time between the first and second messages. I was monitoring with an RTL SDR, so I saw when the meter was replying but it hasn't read the values from the answer correctly did not stop the frequency cycle. The high end causes that the meter replies on a full bandwidth spectrum I believe, otherwise, it just replies around the frequency of the wake up signal. Finding a single frequency is absolutely enough. The original purpose was the same, just hit a right frequency, the tolerance is huge.

xekil, 2022-06-18:
Thank you for your answers, I will continue in English then :)

For the time of the interrogation, I therefore put UTC2 which makes 10 hours in France, I will see on Monday.
Regarding the frequency, I reduced to 433.8615 we will see over several days if it is reliable.

For the addition of an MQTT sensor, I will document myself because I am not sure to understand if it is necessary to add code in the ESP.

Patbob86, 2022-06-18:
Hi everybody!
Thanks for this project I am working on for a few days, unfortunately without success, as I can't get any answer from my Water Counter. 
I use Arduino IDE and I have setup my parameters (wifi, Mqtt, year and Counter number without the first zeros).
I tried a lot of times to connect and scan using the piece of code to find the right frequency in everblu-meters-esp8266.ino, but absolutely no answers at all..
I tried many times, at small distance from the counter (1 meter, 10 meters....). 
Maybe my Counter is briken ??!

I use a D1 Mini and the CC1101 module is a E07-M1101D-SMA. 
The connections between the M1101 module and the D1 Mini seem ok, because I can monitor the sound of the transmission on a Walky Talky according to the frequency displayed on the terminal during the scan.
I don't have a RTL SDR radio module to listen the spectrum.

Is there a specific setup for the M1101D card?
In the different other projects I could find, there is a different value for the GDO0.. 
Some of them say
#define GDO0 0 //header  11
and other, like psykokwak, say
#define GDO0 5 //header  11 
What is the differences in this setting? I tried both, but nothing better...
Merci pour votre aide!

GSzabados, 2022-06-18:
> Some of them say
> #define GDO0 0 //header 11
> and other, like psykokwak, say
> #define GDO0 5 //header 11
> What is the differences in this setting? I tried both, but nothing better...

That is telling the Wemos D1 that which GPIO pin is connected to the GDO0 of the CC1101.

Please open a separate issue, state what is the model of your meter and what is the serial debug showing when it is unable to read the meter. And please state as well when did you try to communicate with the meter.

Patbob86, 2022-06-18:
Hi. Thanks for reply.
Ok I will open an other issue to describe my counter..

> That is telling the Wemos D1 that which GPIO pin is connected to the GDO0 of the CC1101.
> 
Sorry, but I don't understand what you mean..
The GDO0 of the CC1101 (M1101D) is pin 3. 
And it is connected to the Wemos D1Mini to pin D1 as recommended .. so I don't see the meaning of 0 or 5 in the setting  of GD0:
With my wiring, do I put #define GDO0 0 or #define GDO0 5 ?

GSzabados, 2022-06-18:
> Sorry, but I don't understand what you mean..

Open another issue, and I will explain it there.

closed 2022-06-19 by xekil