psykokwak-com/everblu-meters-esp8266 #3: > Sorry, but I don't understand what you mean..

Patbob86, 2022-06-18:
> Sorry, but I don't understand what you mean..

Open another issue, and I will explain it there.

_Originally posted by @GSzabados in https://github.com/psykokwak-com/everblu-meters-esp8266/issues/2#issuecomment-1159484088_

Patbob86, 2022-06-18:
About my counter. 
It is a "classical" Itron EverBlu Cyble Enhanced V2.1 with manufacture date 2014.
Phot attached..
[image]

GSzabados, 2022-06-18:
<html>
<body>
<p dir="auto" style="box-sizing: border-box; margin-top: 0px; margin-bottom: 16px; color: rgb(36, 41, 47); font-family: -apple-system, BlinkMacSystemFont, &quot;Segoe UI&quot;, Helvetica, Arial, sans-serif, &quot;Apple Color Emoji&quot;, &quot;Segoe UI Emoji&quot;; font-size: 16px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; background-color: rgb(255, 255, 255); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;">Pins wiring for<span> </span><a href="https://www.wemos.cc/en/latest/d1/index.html" rel="nofollow" style="box-sizing: border-box; background-color: transparent; color: var(--color-accent-fg); text-decoration: none; transition: color 80ms cubic-bezier(0.33, 1, 0.68, 1) 0s, background-color, box-shadow, border-color;">Wemos D1 board</a><span> </span>:</p>

CC1101 | Wemos
-- | --
VCC | 3V3
GOD0 (GDO0) | D1
CSN (SPI chip select) | D8
SCK (SPI clock) | D5
MOSI (SPI MOSI) | D7
GOD1 (SPI MISO) | D6
GOD2 (GDO2) | D2
GND (ground) | G

</body>

[image]

If you look at the image you can see that the D1 pin of the Wemos D1 Mini is GPIO5

The code defines GPIO5 (the `//header 11` part I believe is a leftover from the previous Rapsberry Pi project, so it is not for the Wemos D1 Mini)

`#define GDO0 5 //header  11 `

GSzabados, 2022-06-18:
So, the next question, when did you try to communicate with the meter?

And can you show what the serial log shows?

Patbob86, 2022-06-18:
OOOKK for the GDO0 !! Thanks for the explanation! I understand now. And I am well with #define GDO0 5...
I have tried to communicate  every day last week at different times between 10 and 18 h. 
For the tests, the module is placed at around 1 meter from the counter... I tried also 10  or 15 meters...
I can monitor the radio with a Walky Talky and I hear the frequence of 1200Hz.. so it seems to send something..
I have switched ON the "scan" mode in the code to scan the band from 433.760 up to 433.890 MHz with steps of 500 hertz
" for (float i = 433.76f; i < 433.890f; i += 0.0005f) { "
It scan all the frequencies one by one with no results, 
Then it connects to wifi, to mqtt and sends the HA frames.. it works, I receive the frame in my Broker. 
And then nothing more..
See printscreen attached.
[image]
.

GSzabados, 2022-06-18:
In the cc1101.cpp change this in Line 24:

`uint8_t debug_out = 0;`

to

`uint8_t debug_out = 1;`

Flash again the ESP and check what is the message when it is testing the frequencies.

You can increase the step to 2000Hz. To find a suitable frequency quicker.

Patbob86, 2022-06-18:
Here is the small part of result because it is a long file!!.. 
Most of the time, it is:
GDO0! (0, 0) TMO on REC
But around 433.778 (or also 433.774), it says:
size_byte=124  l_radian_frame_size_byte=171
GDO0! (0, 0) 1st synch received (1)  rssi=150 lqi=128 F_est=0 
GDO0! (1, 565) TMO on REC
Test frequency : 433.778137
Does it mean that it detects some answer of the counter?
If you wish, I can send you the entire file...
-----

size_byte=18  l_radian_frame_size_byte=25
GDO0! (0, 0) TMO on REC

size_byte=124  l_radian_frame_size_byte=171
GDO0! (0, 0) TMO on REC
Test frequency : 433.772095
MARCSTATE : raw:0x13  0x13 free_byte:0x0F sts:0x02 sending 2s WUP...
202 free_byte:0x0E sts:0x07

size_byte=18  l_radian_frame_size_byte=25
GDO0! (0, 0) TMO on REC

size_byte=124  l_radian_frame_size_byte=171
GDO0! (0, 0) 1st synch received (1)  rssi=149 lqi=128 F_est=0 
GDO0! (1, 400) TMO on REC
Test frequency : 433.774109
MARCSTATE : raw:0x13  0x13 free_byte:0x0F sts:0x02 sending 2s WUP...
202 free_byte:0x0E sts:0x07

size_byte=18  l_radian_frame_size_byte=25
GDO0! (0, 0) TMO on REC

size_byte=124  l_radian_frame_size_byte=171
GDO0! (0, 0) TMO on REC
Test frequency : 433.776123
MARCSTATE : raw:0x13  0x13 free_byte:0x0F sts:0x02 sending 2s WUP...
202 free_byte:0x0E sts:0x07

size_byte=18  l_radian_frame_size_byte=25
GDO0! (0, 0) TMO on REC

size_byte=124  l_radian_frame_size_byte=171
GDO0! (0, 0) 1st synch received (1)  rssi=150 lqi=128 F_est=0 
GDO0! (1, 565) TMO on REC
Test frequency : 433.778137
MARCSTATE : raw:0x13  0x13 free_byte:0x0F sts:0x02 sending 2s WUP...
202 free_byte:0x0E sts:0x07

size_byte=18  l_radian_frame_size_byte=25
GDO0! (0, 0) TMO on REC

size_byte=124  l_radian_frame_size_byte=171
GDO0! (0, 0) TMO on REC
Test frequency : 433.780151
MARCSTATE : raw:0x13  0x13 free_byte:0x0F sts:0x02 sending 2s WUP...
202 free_byte:0x0E sts:0x07

GSzabados, 2022-06-18:
Enable this in the cc1101.cpp at line 793:

[`//delay(30); //43ms de bruit`](https://github.com/psykokwak-com/everblu-meters-esp8266/blob/main/cc1101.cpp#L793)

by removing the `//`

And try again the frequency scan.

Patbob86, 2022-06-18:
With putting this delay(30), it has found only 1 RSSI at 433.800..

size_byte=18  l_radian_frame_size_byte=25
GDO0! (0, 0) TMO on REC

size_byte=124  l_radian_frame_size_byte=171
GDO0! (0, 0) TMO on REC
Test frequency : 433.798279
MARCSTATE : raw:0x13  0x13 free_byte:0x0F sts:0x02 sending 2s WUP...
202 free_byte:0x0E sts:0x07

size_byte=18  l_radian_frame_size_byte=25
GDO0! (0, 0) TMO on REC

size_byte=124  l_radian_frame_size_byte=171
GDO0! (0, 0) 1st synch received (1)  rssi=152 lqi=128 F_est=254 
GDO0! (1, 250) TMO on REC
Test frequency : 433.800293
MARCSTATE : raw:0x13  0x13 free_byte:0x0F sts:0x02 sending 2s WUP...
202 free_byte:0x0E sts:0x07

size_byte=18  l_radian_frame_size_byte=25
GDO0! (0, 0) TMO on REC

size_byte=124  l_radian_frame_size_byte=171
GDO0! (0, 0) TMO on REC
Test frequency : 433.802307
MARCSTATE : raw:0x13  0x13 free_byte:0x0F sts:0x02 sending 2s WUP...
202 free_byte:0x0E sts:0x07

size_byte=18  l_radian_frame_size_byte=25
GDO0! (0, 0) TMO on REC

size_byte=124  l_radian_frame_size_byte=171
GDO0! (0, 0) TMO on REC
Test frequency : 433.804321
MARCSTATE : raw:0x13  0x13 free_byte:0x0F sts:0x02 sending 2s WUP...
202 free_byte:0x0E sts:0x07

size_byte=18  l_radian_frame_size_byte=25
GDO0! (0, 0) TMO on REC

GSzabados, 2022-06-18:
Oh, wait a second. Are you trying today? Today is Saturday! It will not reply until Monday!

Monday to Friday... Working hours...

Patbob86, 2022-06-18:
Aaaaaaaahhhh YESSSS!! You are perfectly right!! It is week-end for the counters too :-)
I'll continue the tests and let you know on next Monday..
Many thanks for your help until now ! Have a nice week-end.. here in France, still 39°C outside..

Patbob86, 2022-06-20:
Hi!
I have done many tests today Monday (it's a working day, so the counter should answer !)
I tried to setup with or without delay(30); //43ms de bruit, and I have put the boards and my computer at around 2 meters from the counter.
Results are so disappointing as the other times .. Absolutely no stable answer from the counter. Sometimes, some RSSI detected, but never at the same frequency.
Examples:
size_byte=124  l_radian_frame_size_byte=171
GDO0! (0, 0) 1st synch received (1)  rssi=150 lqi=128 F_est=253 
GDO0! (1, 650) TMO on REC
Test frequency : 433.794250
------
size_byte=124  l_radian_frame_size_byte=171
GDO0! (0, 0) 1st synch received (1)  rssi=155 lqi=128 F_est=0 
GDO0! (1, 495) TMO on REC
Test frequency : 433.822449
MARCSTATE : raw:0x13  0x13 free_byte:0x0F sts:0x02 sending 2s WUP...
202 free_byte:0x0E sts:0x07
-------
size_byte=124  l_radian_frame_size_byte=171
GDO0! (0, 0) 1st synch received (1)  rssi=153 lqi=128 F_est=255 
GDO0! (1, 615) TMO on REC
Test frequency : 433.814392
MARCSTATE : raw:0x13  0x13 free_byte:0x0F sts:0x02 sending 2s WUP...
202 free_byte:0x0E sts:0x07
----
I really don't know what to do now!!

Patbob86, 2022-06-20:
Thanks for the reply!
Yes, the counter is a 433Mhz (written on it.. see the photo at the top of this thread)
With the Talky, I hear only the "long" bip around 2 secondes. Never heard any short signal back after the long one..
I have not  a RTL SDR and would not like to invest 40 euros just to use it once and then put it in a drawer !

GSzabados, 2022-06-20:
> Yes, the counter is a 433Mhz

I meant the CC1101 is 433Mhz? There are different models.

Patbob86, 2022-06-20:
Anybody knows what is the meaning of this 3 words during the measurement :
rssi=150 lqi=128 F_est=253
I thought that rssi is the signal strengh given in dBm ?? but this value is not correct because 1 mW = 0 dBm and a Cyble RF counter has an output power max of 10 mW wich is = 10dBm.... not 150!
And what means lqi and F_Est ?

pg68390, 2022-06-20:
Bonjour, je suis novice et j'aurais besoin d'aide, j'ai suivi les indications de Jean Charles qui se trouvent dans le readme, mais lorsque je compile le sketch, j'ai un message d'erreur : arduino.h: No such file or directory, pouvez vous m'aider à résoudre ce problème ? merci cordialement

Patbob86, 2022-06-20:
Bonjour. Perso, je n'ai eu aucun souci de compilation en utilisant la distribution  arduino-1.8.19 et en faisant toutes les mises à jour ESP32 et ESP8266 avant compilation..
La compil se passe parfaitement bien..

pg68390, 2022-06-21:
Merci Patbob86 pour la réponse rapide, j'ai effectivement la même version d'Arduino 1.8.19 mais je reçois ce message d'erreur lors de la compilation, par contre si je modifie l'arduino.h en Arduino.h avec une A majuscule, çà passe mais je ne sais pas si celà compromet le fonctionnement du projet ?

Patbob86, 2022-06-21:
Si c'est ok avec un A majuscule, c'est que ton script qui appelle arduino.h contient #include <Arduino.h> car les majuscules sont gérées. 
Par contre pour la compil qui nous intéresse d'Everblu, 3 scripts contiennent un #include <arduino.h>. 
Ce sont cc1101.cpp, utils.cpp et utils.h.. et aucun d'entre eux n'a de A majuscule..
Sinon, si la compil passe avec un A majuscule, ça ne doit pas changer grand chose une fois compilé..
C'est très curieux ton affaire! il faudrait trouver d'où vient cet appel avec un A majuscule..

pg68390, 2022-06-21:
@Patbob86 Merci encore de prendre le temps de me répondre, le fait est que j'ai fait une recherche sur mon pc de ce fameux <arduino.h> parmi les librairies déjà installées et j'ai remarqué que j'avais à plusieurs endroits ce <Arduino.h> avec la majuscule, donc j'ai remplacé dans les 3 scripts le a par A dans le sketch pour voir si çà passait mais je ne savais pas que les majuscules étaient gérées et donc mon Arduino.h avec majuscule n'a sans doute rien à voir avec ce que veut notre croquis !
Je dois encore creuser cette affaire...

Patbob86, 2022-06-21:
Je ne m'étais jamais penché sur la question, mais effectivement, ya un truc bizarre.. En faisant une recherche dans les fichiers de mon directory arduino-1.8.19, je trouve un Arduino.h dans 
D:\arduino-1.8.19\portable\packages\arduino\hardware\avr\1.8.5\cores\arduino qui a une taille de 7.43 Ko, un arduino.h dans D:\arduino-1.8.19\portable\packages\arduino\hardware\avr\1.8.5\firmwares\wifishield\wifi_dnld\src\SOFTWARE_FRAMEWORK\BOARDS\ARDUINO qui fait 8.48 Ko et encore un autre arduino.h dans D:\arduino-1.8.19\portable\packages\arduino\hardware\avr\1.8.5\firmwares\wifishield\wifiHD\src\SOFTWARE_FRAMEWORK\BOARDS\ARDUINO qui fait 8.60 Ko!!
Donc, noms identiques, mais certainement versions différentes?!
C'est un bazar la dedans!! reste à savoir lequel est utilisé par la compil Everblu??! Peut-être pour ça que chez moi, ça ne fonctionne pas une compil sans erreur..?

pg68390, 2022-06-22:
Effectivement c'est le bazar, j'ai un 2ème PC portable sous W10 et là la compilation se fait sans soucis après l'ajout de la bibliothèque EspMQTTClient avec une version d'Arduino plus ancienne 1.8.13 ! Pour info, sur le PC où j'ai le souci je suis sur Linux (Zorin) avec l'arduino 1.8.19.
J'ai hâte de recevoir mon CC1101 pour pouvoir tester en "vrai" ce sketch en espérant ne pas rencontrer d'autres problèmes.

Patbob86, 2022-06-22:
Pour info, je viens de faire de nouveaux essais avec un autre module radio CC1101 qu'on m'a prêté. J'ai exactement les mêmes résultats qu'avec le mien.. aucune réponse du compteur.. cela élimine donc une défaillance du module..
Reste la  possibilité de la radio du  compteur qui ne fonctionne plus, ou du Firmware qui se compile avec des librairies qui ne sont pas de la même version que celles utilisées par l'auteur Psykokwak..?
Ce serait intéressant que les personnes qui ont réussi à faire fonctionner tout ça nous disent quels outils et versions ils ont utilisé.. Malheureusement, il n'y a pas foule pour nous épauler :-(
@GSzabados, if you still follow this thread, coud you tell us what tools you use to compil the Sketch and if all is running fine for you?

xekil, 2022-06-22:
Bonjour, pour ma part j'ai utlisé Arduino IDE 1.8.19 pour Windows (64bits)
J'ai ensuite ajouter la librairie pour ESP et pas de souci de compilation, le test à fonctionné du premier coup en étant à 2 mètres du compteur.

GSzabados, 2022-06-22:
> coud you tell us what tools you use to compil the Sketch

I will have a look at it. I was not really following your conversation, but definitely there shouldn't be any issue to compile it to an ESP32, as I used one.

Patbob86, 2022-06-22:
> I will have a look at it. I was not really following your conversation, but definitely there shouldn't be any issue to compile it to an ESP32, as I used one.
Yes @GSzabados, you are right.. Me too, I have no issue during compile.. but counter always mute... so it is why I try to investigate on the side of compil and libraries .
But me, its is not ESP32 but ESP8266 (Wemos D1mini).
Small question: in the choice of the board in the Arduino IDE menu, which D1mIni do you choose? Me I have used LOLIN (WEMOS) D1R2Mini.. Is it the same choice with you?
[image]

GSzabados, 2022-06-22:
> Small question: in the choice of the board in the Arduino IDE menu, which D1mIni do you choose? Me I have used LOLIN (WEMOS) D1R2Mini.. Is it the same choice with you?

That should be the right one. As the code compiles, and it seems to behave correctly with the code, I don't expect that you have chosen it wrongly.

GSzabados, 2022-06-22:
So, back to some other questions, I am not sure that have I asked it before or not.

Has anybody physically reading your meter to get your readings and issue your a bill? If not, then the Smart Meter should be working.

I have a 2019 model, it has still 142 months lifetime. I wouldn't expect from a 2014 model, that it has ran out of battery already.

Do you have the right serial number and year of production (14) added at the end everblu_meter.c?

Serial number should be 7 digits, except when the serail starts with a 0, then you should just put there the last 6 digit without 0.

The serial is exactly just the middle part, what you have blanked out on your picture.

Patbob86, 2022-06-22:
>Has anybody physically reading your meter to get your readings and issue your a bill? If not, then the Smart Meter should be working.
The reader of the counter by the company is once a year and I think that last time was around september 2021.. so, maybe you are right and the battery is Out...
>Do you have the right serial number and year of production (14) added at the end everblu_meter.c?
Yes I thing it is correct:
#define METER_YEAR 14
#define METER_SERIAL 808889
the real number is 0808889  (and I have removed the left zéro digit)
So, as you can see, all seems well setup and should logicaly work... but it does not!! It is a big headhake!!

GSzabados, 2022-06-22:
> The reader of the counter by the company is once a year and I think that last time was around september 2021.. so, maybe you are right and the battery is Out...

Are you billed once every year, or every month / second month? Or are you paying an average after the last year's consumption monthly and adjusted after the reading?
Just asking to verify, that meter is working or not, with some logic.

My meter is read every second month, it is in my cellar. A car drives around with an antenna on the top and reads the meters remotely. The company hasn't even came into my house to read it physically since it was installed. They bill me every second month as well.

The values seems to be fulfilled correctly. Yes, I meant h, looked at the wrong file regarding the ending.

You can try to enlarge the range of frequency for the search. It can be that the CC1101 is highly inaccurate.

Also it is possible that the meter replies, but the device is unable to decode it correctly due to timing issues. And due to that, it doesn't lock on the right frequency. 
When I was trying my setup, I had the RTL SDR and I could see when the meter replies, and I could see, that the code is unable to decode the message and keeps scanning the frequencies. In my case it was the timing. When I added that 30ms delay, it was able to decode the message correctly.

Patbob86, 2022-06-22:
I pay 2 times per year every 6 months. 1 with estimated consumption and 1 with real index of the counter . Bill is adjusted and calculated according to match the real index and the estimated consumtion already paid. Index looks match the reality..
I see that the adjustments are not easy at all. Some guys are very lucky and it works at first power on, but for the others, as we say in french "bonjour la galère"!
I will try to check with the counter of my neighbour in the next days.. maybe more lucky?? But for no, I give up a little and let the other "colleagues" test and tes again :-)

GSzabados, 2022-06-22:
> I pay 2 times per year every 6 months. 1 with estimated consumption and 1 with real index of the counter . Bill is adjusted and calculated according to match the real index and the estimated consumtion already paid. Index looks to match the reality..

I have a feeling then that your smart meter has never been commissioned. And was never working. The point of these smart meters that you don't have to enter the building to read it. You can do it remotely from the street.

The company must have just installed the smart meter, but they never used it. A waste of money then...

Maybe asked the provider that how is the smart meter working? Or are they using it at all... 
You might find out the correct answer for your problem...

GSzabados, 2022-06-22:
By the way, you could just order an RTL SDR from Amazon, try it, and then send it back, that it was not working for you... ;)

Patbob86, 2022-06-22:
"By the way, you could just order an RTL SDR from Amazon, try it, and then send it back, that it was not working for you... ;)"
Why not ??!!!

GSzabados, 2022-06-22:
I used this to visualize the signal:

https://airspy.com/

Cheaper versions of the RTLSDR would work as well, like the NESDR.

Look at the comments here regarding that:
http://www.lamaisonsimon.fr/wiki/doku.php?id=maison2:compteur_d_eau:compteur_d_eau

Patbob86, 2022-06-22:
Thanks for these infos! I'll see what I do..

smartmatic, 2022-07-09:
I am facing the same issue that i don't get the data into Home Assistant. But i think i find out the right frequency?
For one frequency i get this result:

`20:21:12.874 -> 3ffff9a0:  00000002 3fff0001 3fff0064 4020106f  
20:21:12.874 -> 3ffff9b0:  3ffe8630 00000096 00000064 40201124  
20:21:12.874 -> 3ffff9c0:  3ffe0000 00000096 00000064 402010e0  
20:21:12.874 -> 3ffff9d0:  00000000 00000001 3fff0064 402014db  
20:21:12.874 -> 3ffff9e0:  3ffffa00 3fff0001 00000008 40201686  
20:21:12.874 -> 3ffff9f0:  00000064 3ffffa20 00000000 402010e0  
20:21:12.912 -> 3ffffa00:  00000000 3ffe8630 3fffff60 3fffff60  
20:21:12.912 -> 3ffffa10:  0000004c 3ffffed0 00000000 40201977  
20:21:12.912 -> 3ffffa20:  feefeffe feefeffe feefeffe feefeffe  
20:21:12.912 -> 3ffffa30:  feefeffe feefeffe feefeffe feefeffe  
20:21:12.912 -> 3ffffa40:  feefeffe feefeffe feefeffe feefeffe  
20:21:12.912 -> 3ffffa50:  feefeffe feefeffe feefeffe feefeffe  
20:21:12.912 -> 3ffffa60:  feefeffe feefeffe feefeffe feefeffe  
20:21:12.912 -> 3ffffa70:  feefeffe feefeffe feefeffe feefeffe  
20:21:12.912 -> 3ffffa80:  feefeffe feefeffe feefeffe feefeffe  
20:21:12.946 -> 3ffffa90:  feefeffe feefeffe feefeffe feefeffe  
20:21:12.946 -> 3ffffaa0:  feefeffe feefeffe feefeffe feefeffe  
20:21:12.946 -> 3ffffab0:  feefeffe feefeffe feefeffe feefeffe  
20:21:12.946 -> 3ffffac0:  feefeffe feefeffe feefeffe feefeffe  
20:21:12.946 -> 3ffffad0:  feefeffe feefeffe feefeffe feefeffe  
20:21:12.946 -> 3ffffae0:  feefeffe feefeffe feefeffe feefeffe  
20:21:12.946 -> 3ffffaf0:  feefeffe feefeffe feefeffe feefeffe  
20:21:12.946 -> 3ffffb00:  feefeffe feefeffe feefeffe feefeffe  
20:21:12.985 -> 3ffffb10:  feefeffe feefeffe feefeffe feefeffe  
20:21:12.985 -> 3ffffb20:  feefeffe feefeffe feefeffe feefeffe  
20:21:12.985 -> 3ffffb30:  feefeffe feefeffe feefeffe feefeffe  
20:21:12.985 -> 3ffffb40:  feefeffe feefeffe feefeffe feefeffe  
20:21:12.985 -> 3ffffb50:  feefeffe feefeffe feefeffe feefeffe  
20:21:12.985 -> 3ffffb60:  feefeffe feefeffe feefeffe 3ffe83c4  
20:21:12.985 -> 3ffffb70:  4021d745 0000001c 00000020 40100d14  
20:21:12.985 -> 3ffffb80:  40100f82 feefeffe 00000001 40100f74  
20:21:12.985 -> 3ffffb90:  feefeffe 0000001c 00000001 40212928  
20:21:13.014 -> 3ffffba0:  00000002 00000001 3ffe83c4 3ffe83c4  
20:21:13.014 -> 3ffffbb0:  4021d745 00000018 00000020 40100d14  
20:21:13.014 -> 3ffffbc0:  40100f82 feefeffe 00000001 40100f74  
20:21:13.014 -> 3ffffbd0:  3ffffc04 00000018 00000001 40212928  
20:21:13.014 -> 3ffffbe0:  00000001 7fcb4000 3ffe83c4 4021d745  
20:21:13.014 -> 3ffffbf0:  3fff0efc 3fff0efd 00000009 4021cac9  
20:21:13.047 -> 3ffffc00:  fffffff4 00000015 feefeffe feefeffe  
20:21:13.047 -> 3ffffc10:  00000002 00000002 00000009 3fff0ed4  
20:21:13.047 -> 3ffffc20:  44e20000 3e975b9d 3fff0ef4 0000000e  
20:21:13.047 -> 3ffffc30:  00000000 00000003 00000008 00000000  
20:21:13.047 -> 3ffffc40:  00000000 00000006 00000000 407b1c29  
20:21:13.047 -> 3ffffc50:  00000002 00000000 00000000 feefeffe  
20:21:13.047 -> 3ffffc60:  00000009 3fe88000 112a8800 3ffe83c4  
20:21:13.047 -> 3ffffc70:  00000000 00000000 3ffffe10 4021e268  
20:21:13.087 -> 3ffffc80:  3fff0ef7 3ffffda3 00000000 40217579  
20:21:13.087 -> 3ffffc90:  3ffffd08 3ffffca4 3ffffca0 3ffe83c4 <
20:21:13.087 -> 3ffffca0:  00000006 3ffffd60 3ffffe10 4021e268  
20:21:13.087 -> 3ffffcb0:  4021e1a4 3fff0ef7 00000006 3fff0efd  
20:21:13.087 -> 3ffffcc0:  feefeffe feefeffe feefeffe 00000006  
20:21:13.087 -> 3ffffcd0:  3fff0ef4 3ffffd60 3ffe83c4 40217abd  
20:21:13.087 -> 3ffffce0:  3ffffd0c 3ffffd08 00000066 3ffffd04  
20:21:13.087 -> 3ffffcf0:  00000000 feefeffe feefeffe feefeffe  
20:21:13.087 -> 3ffffd00:  3ffe8e2f 00000009 00000003 feefef00  
20:21:13.119 -> 3ffffd10:  3ffffe10 4021e1a4 00000066 3ffe83c4  
20:21:13.119 -> 3ffffd20:  00000001 00000000 3ffffe10 4021e268  
20:21:13.119 -> 3ffffd30:  ffffffff 3ffe8e2f 00000001 feefeffe  
20:21:13.119 -> 3ffffd40:  feefeffe feefeffe feefeffe 3ffe8e2f  
20:21:13.119 -> 3ffffd50:  3ffe8e30 00000000 3ffffe10 4021e469  
20:21:13.119 -> 3ffffd60:  00000400 00000006 00000000 00000000  
20:21:13.119 -> 3ffffd70:  0000000a 0000001c fe302066 feefeffe  
20:21:13.119 -> 3ffffd80:  feefeffe feefeffe feefeffe feefeffe  
20:21:13.152 -> 3ffffd90:  feefeffe feefeffe feefeffe feefeffe  
20:21:13.152 -> 3ffffda0:  00efeffe feefeffe 00000000 407b1c29  
20:21:13.152 -> 3ffffdb0:  feefeffe feefeffe 00000003 feefeffe  
20:21:13.152 -> 3ffffdc0:  3fffff40 3fffff10 00000010 feefeffe  
[... 30 lines cut ...]
20:21:13.294 -> 3fffffb0:  feefeffe feefeffe 3ffe8654 40100fa5  
20:21:13.294 -> <<<stack<<<
20:21:13.294 -> 
20:21:13.294 -> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
20:21:13.331 -> 
20:21:13.331 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
20:21:13.331 -> 
20:21:13.331 -> load 0x4010f000, len 3460, room 16 
20:21:13.331 -> tail 4
20:21:13.331 -> chksum 0xcc
20:21:13.331 -> load 0x3fff20b8, len 40, room 4 
20:21:13.331 -> tail 4
20:21:13.331 -> chksum 0xc9
20:21:13.331 -> csum 0xc9
20:21:13.331 -> v0005af30
20:21:13.331 -> ~ld
20:21:13.405 -> 
20:21:13.405 -> 
20:21:13.405 -> Test frequency : 433.760010
20:21:13.405 -> MARCSTATE : raw:0x00  0x00 free_byte:0x00 sts:0x00 sending 2s WUP...
20:21:13.405 -> 0 free_byte:0x00 sts:0x00
20:21:13.405 -> 
20:21:13.405 -> size_byte=18  l_radian_frame_size_byte=25`

What does it mean?

smartmatic, 2022-07-09:
In addition i tried an ESP32-C3-DEVKITM-1

Here is the result:

20:35:10.179 -> ⸮  ⸮  ⸮  ⸮  ⸮  ⸮  ⸮  ⸮  ⸮  ⸮  ⸮  ⸮  ⸮  ⸮  ⸮  ⸮  ⸮  ⸮  ⸮  ⸮  ⸮  ⸮ESP-ROM:esp32c3-api1-20210207
20:38:04.068 -> Build:Feb  7 2021
20:38:04.068 -> rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
20:38:04.068 -> SPIWP:0xee
20:38:04.068 -> mode:DIO, clock div:1
20:38:04.068 -> load:0x3fcd6100,len:0x438
20:38:04.095 -> load:0x403ce000,len:0x918
20:38:04.095 -> load:0x403d0000,len:0x24e4
20:38:04.095 -> SHA-256 comparison failed:
20:38:04.095 -> Calculated: 080c5cb68a075ced55f248b97bca965e3e5bd5da80a64e34e6a1638f89d6f64e
20:38:04.095 -> Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
20:38:04.095 -> Attempting to boot anyway...
20:38:04.095 -> entry 0x403ce000
20:38:04.234 -> 
20:38:04.234 -> 
20:38:04.234 -> Test frequency : 433.760010
20:38:04.268 -> MARCSTATE : raw:0xFF  0x1F free_byte:0x00 sts:0x0F sending 2s WUP...
20:38:04.268 -> 0 free_byte:0x00 sts:0x0F
20:38:04.268 -> 
20:38:04.268 -> size_byte=18  l_radian_frame_size_byte=25
20:38:04.952 -> GDO0! (0, 1) 1st synch received (127)  rssi=182 lqi=255 F_est=255 
20:38:04.989 -> GDO0! (1, 12) frame received (127)
20:38:04.989 -> 
20:38:04.989 -> size_byte=124  l_radian_frame_size_byte=171
20:38:05.133 -> GDO0! (0, 13) 1st synch received (127)  rssi=182 lqi=255 F_est=255 
20:38:05.204 -> GDO0! (1, 36) frame received (762)
20:38:05.275 -> stop bit error10Test frequency : 433.760498
20:38:05.275 -> MARCSTATE : raw:0xFF  0x1F free_byte:0x0F sts:0x0F sending 2s WUP...
20:38:05.275 -> 0 free_byte:0x0F sts:0x0F
20:38:05.275 -> 
20:38:05.275 -> size_byte=18  l_radian_frame_size_byte=25
20:38:58.233 -> ESP-ROM:esp32c3-api1-20210207
20:38:58.269 -> Build:Feb  7 2021
20:38:58.269 -> rst:0x1 (POWERON),boot:0xd (SPI_FAST_FLASH_BOOT)
20:38:58.269 -> SPIWP:0xee
20:38:58.269 -> mode:DIO, clock div:1
20:38:58.269 -> load:0x3fcd6100,len:0x438
20:38:58.269 -> load:0x403ce000,len:0x918
20:38:58.269 -> load:0x403d0000,len:0x24e4
20:38:58.269 -> SHA-256 comparison failed:
20:38:58.269 -> Calculated: 080c5cb68a075ced55f248b97bca965e3e5bd5da80a64e34e6a1638f89d6f64e
20:38:58.269 -> Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
20:38:58.269 -> Attempting to boot anyway...
20:38:58.307 -> entry 0x403ce000
20:38:58.416 -> 
20:38:58.416 -> 
20:38:58.416 -> Test frequency : 433.760010
20:38:58.451 -> MARCSTATE : raw:0x00  0x00 free_byte:0x0F sts:0x0F sending 2s WUP...
20:38:58.451 -> 0 free_byte:0x0F sts:0x0F
20:38:58.451 -> 
20:38:58.451 -> size_byte=18  l_radian_frame_size_byte=25
20:39:05.672 -> ESP-ROM:esp32c3-api1-20210207
20:39:05.672 -> Build:Feb  7 2021
20:39:05.672 -> rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
20:39:05.672 -> SPIWP:0xee
20:39:05.672 -> mode:DIO, clock div:1
20:39:05.672 -> load:0x3fcd6100,len:0x438
20:39:05.672 -> load:0x403ce000,len:0x918
20:39:05.672 -> load:0x403d0000,len:0x24e4
20:39:05.672 -> SHA-256 comparison failed:
20:39:05.707 -> Calculated: 080c5cb68a075ced55f248b97bca965e3e5bd5da80a64e34e6a1638f89d6f64e
20:39:05.707 -> Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
20:39:05.707 -> Attempting to boot anyway...
20:39:05.707 -> entry 0x403ce000
20:39:05.852 -> 
20:39:05.852 -> 
20:39:05.852 -> Test frequency : 433.760010
20:39:05.852 -> MARCSTATE : raw:0xFF  0x1F free_byte:0x0F sts:0x0F sending 2s WUP...
20:39:05.852 -> 0 free_byte:0x0F sts:0x0F
20:39:05.852 -> 
20:39:05.852 -> size_byte=18  l_radian_frame_size_byte=25
20:39:05.890 -> GDO0! (0, 8) 1st synch received (56)  rssi=142 lqi=231 F_est=51 
20:39:11.016 -> GDO0! (1, 27) frame received (127)
20:39:11.016 -> 
20:39:11.016 -> size_byte=124  l_radian_frame_size_byte=171
20:39:11.016 -> GDO0! (0, 4) 1st synch received (127)  rssi=182 lqi=255 F_est=255 
20:39:11.087 -> GDO0! (1, 24) frame received (758)
20:39:11.200 -> stop bit error10Test frequency : 433.760498
20:39:11.200 -> MARCSTATE : raw:0xFF  0x1F free_byte:0x0F sts:0x0F sending 2s WUP...
20:39:11.200 -> 0 free_byte:0x0F sts:0x0F
20:39:11.200 -> 
20:39:11.200 -> size_byte=18  l_radian_frame_size_byte=25
`
`

GSzabados, 2022-07-09:
@smartmatic, have you tried it just literally now?

smartmatic, 2022-07-09:
> have you tried it just literally now

Sorry i dont understand your question?
But yes i tried it in the last hour. 
I know ist weekend but in your documentation is mentioned that it can be that some meters are working only during working days. Means other meters sending every day data right?

GSzabados, 2022-07-09:
@smartmatic 

I don't want to be rude, but I think you guys have to read the `read me` gain to understand the function of business hours.

https://github.com/psykokwak-com/everblu-meters-esp8266#business-hours

TLTR: The meter is not going to answer on a Saturday! Neither on a Sunday! And generally outside of business hours!

smartmatic, 2022-07-09:
> https://github.com/psykokwak-com/everblu-meters-esp8266#business-hours

But you are writing "Your meter may be configured in such a way..."
May means can be or can not be.

GSzabados, 2022-07-09:
> that some meters are working only during working days

If you would be a business, and you would invest into something to consume power when you are sure that you are not going to use it, would you keep it on for that period? 

It is an easy economical question. That is one of the most brilliant power saving techniques for any device which would be queried during normal human business hours, when the guy who supposed to read it would roam around the streets. No other times.

Try it during business hours! That is the point. Don't assume that it will work. Because it will not... 

Do you have an RTLSDR to confirm that is it replying? If yes, and you can see that it is replying then please carry on...

smartmatic, 2022-07-09:
I don't have an RTLSDR. Can you recommend please a device from amazon?

Please explain what this monitor message says?

20:48:15.629 -> ⸮⸮    ⸮⸮⸮⸮⸮⸮⸮ ⸮⸮  ⸮ ⸮ ⸮⸮⸮ ⸮⸮⸮⸮⸮ ⸮⸮ ⸮ ⸮  ⸮  ⸮ ⸮⸮  ⸮ ⸮⸮ ⸮ ⸮⸮ ⸮⸮  ⸮GDO0! (0, 1) 1st synch received (127)  rssi=182 lqi=255 F_est=255 
20:48:15.684 -> GDO0! (1, 12) frame received (762)
20:48:15.684 -> Test frequency : 433.760498
20:48:15.719 -> MARCSTATE : raw:0xFF  0x1F free_byte:0x0F sts:0x0F sending 2s WUP...
20:48:15.719 -> 0 free_byte:0x0F sts:0x0F
20:48:15.719 -> 
20:48:15.719 -> size_byte=18  l_radian_frame_size_byte=25
`
Basically a sign that it's communicating?
`

GSzabados, 2022-07-09:
If it does for most of the test frequencies, then yes it possible that it communicates. Otherwise it means that it received something, what the code assumes that it is a signal from the meter.

I saw that quite often when I tried to communicate far away from the meter, or during a Saturday/Sunday, but I had an RTLSDR as well, where I could clearly see, that the meter is not replying anything.

You can try that! Change the serial number of the meter in the code to something complete different and check is it showing the same messages.

It is possibly a reflected signal of the previous wake up one, which became distorted.

smartmatic, 2022-07-09:
Ok, will test.
Can you please recommend a RTLSDR on Amazon and shortly described how to use it?
Thank you in advance!
It would be really cool to get it running as your solution is the only one i‘ve found!

GSzabados, 2022-07-09:
> Can you please recommend a RTLSDR on Amazon and shortly described how to use it?

This has everything what you really need at the RTL SDR section.

http://www.lamaisonsimon.fr/wiki/doku.php?id=maison2:compteur_d_eau:compteur_d_eau

It will work, just try it during business hours! My meter comes on actually at 5:49, so maybe working hours from 6:00, but it has shifted a bit due to some wrong time keeping. So you can try it early morning on a Monday...

smartmatic, 2022-07-11:
Good news, the frequency i detected through the serial monitor is the right one. 
I see today a result in the serial monitor!

08:45:54.246 -> ------------------------------
08:45:54.246 -> Got frequency : 433.760010
08:45:54.246 -> ------------------------------
08:45:54.246 -> Liters : 125426
08:45:54.246 -> Battery (in months) : 169
08:45:54.246 -> Counter : 98
08:45:54.283 -> 
08:45:55.989 -> 
08:45:55.989 -> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
08:45:55.989 -> 
08:45:55.989 -> Soft WDT reset
08:45:55.989 -> 
08:45:55.989 -> >>>stack>>>
08:45:55.989 -> 
08:45:55.989 -> ctx: cont
08:45:55.989 -> sp: 3ffffdb0 end: 3fffffc0 offset: 01a0
08:45:55.989 -> 3fffff50:  0000010b 3fff009c 43d8e148 402020e8  
08:45:55.989 -> 3fffff60:  0001e9f2 00000062 000000a9 00000006  
08:45:55.989 -> 3fffff70:  00000012 feefeffe feefeffe feefeffe  
08:45:56.026 -> 3fffff80:  00000062 0001e9f2 feefeffe feefeffe  
08:45:56.026 -> 3fffff90:  feefeffe feefeffe feefeffe 3fff022c  
08:45:56.026 -> 3fffffa0:  3fffdad0 00000000 3fff0218 40211e88  
08:45:56.026 -> 3fffffb0:  feefeffe feefeffe 3ffe8654 40100fa5  
08:45:56.026 -> <<<stack<<<
08:45:56.026 -> 
08:45:56.026 -> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
08:45:56.026 -> 
08:45:56.026 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
08:45:56.065 -> 
08:45:56.065 -> load 0x4010f000, len 3460, room 16 
08:45:56.065 -> tail 4
08:45:56.065 -> chksum 0xcc
08:45:56.065 -> load 0x3fff20b8, len 40, room 4 
08:45:56.065 -> tail 4
08:45:56.065 -> chksum 0xc9
08:45:56.065 -> csum 0xc9
08:45:56.065 -> v0005af30
08:45:56.065 -> ~ld
08:45:56.141 -> 
08:45:56.141 -> 
08:45:56.141 -> Test frequency : 433.760010
08:45:56.141 -> MARCSTATE : raw:0x13  0x13 free_byte:0x0F sts:0x02 sending 2s WUP...
08:45:58.328 -> 203 free_byte:0x0D sts:0x07
08:45:58.328 -> 
08:45:58.328 -> size_byte=18  l_radian_frame_size_byte=25
08:45:58.328 -> GDO0! (0, 0) 1st synch received (1)  rssi=186 lqi=128 F_est=255 
08:45:58.397 -> GDO0! (1, 65) TMO on REC
08:45:58.503 -> 
08:45:58.503 -> size_byte=124  l_radian_frame_size_byte=171
08:45:58.503 -> GDO0! (0, 0) 1st synch received (2)  rssi=186 lqi=128 F_est=255 
08:45:58.612 -> GDO0! (1, 95) frame received (687)
08:45:59.195 -> 
08:45:59.195 -> ------------------------------
08:45:59.195 -> Got frequency : 433.760010
08:45:59.195 -> ------------------------------
08:45:59.195 -> Liters : 125426
08:45:59.195 -> Battery (in months) : 169
08:45:59.195 -> Counter : 99
08:45:59.195 -> 
`
I guess i need to add the exact frequency in the **erverblu_meters.h** in order to get the data into Home Assistant?
`

smartmatic, 2022-07-13:
For whatever reason my d1 is no longer able to connect to WLAN. So i get data from my Cyble Meter but the data is not send to Home Assistant as my d1 does no longer connect to the WLAN. I deleted the d1 IP from my router but the devices does not reconnect to get a new IP assigend.

Please support me!

GSzabados, 2022-07-13:
@smartmatic, you can try to trigger it from Home Assistant. If you have put correctly the MQTT settings, then Home Assistant should have auto discovered it already.

Otherwise, I would check again the WiFi settings (AP and password), and also the MQTT settings in the code, just to make sure you have put the right one.

psykokwak-com, 2022-07-15:
Hi all,
I'm sorry. I left the project.
Feel free to improve it by sending PR or fork.

Patbob86, 2022-07-16:
Bon bin ya plus qu'à tout jeter!! 
plusieurs semaines d'énergie et de temps passé.... 
A mon niveau,  petit test avec modif du code pour envoyer une trame mqtt avec la date/heure du démarrage du D1Mini et de chaque prélèvement des données.. résultat:  le mqtt fonctionne bien, je reçois la trame dans le docker.
Pour moi; MQTT ok .. mais aucune réponse du compteur à mes interrogations..
bye bye! Bon courage à tous!!
tout est à jeter... aucun support ... projet à  oublier immédiatement! 
Dommage :-(

pg68390, 2022-07-16:
@Patbob86 Désolé d'apprendre pour toi que çà ne fonctionne toujours pas, juste pour info, de mon coté après quelques soucis j'ai enfin réussi à recevoir les infos du compteur, j'ai galéré car le schéma de branchement fourni avec le module CC1101 n'était pas le bon, donc je ne comprenais pas pourquoi çà fonctionnait pas, tout est maintenant en ordre et j' enregistre mes relévés dans influxdb et lecture dans grafana.
J'ai vu que le projet n'était plus suivi, dommage. bien cordialement Phil

Patbob86, 2022-07-16:
@pg68390 
Sincèrement content pour toi! 
C'était quoi le branchement du CC1101 pas bon? Et quel branchement a fonctionné pour toi? 
Peut-être que ça pourrait m'aider? ;-)

smartmatic, 2022-07-16:
Hey guys,

is it possible for you to answer in english. Maybe i can also learn from your experience?!

pg68390, 2022-07-16:
@smartmatic  désolé je ne parle pas anglais et je n'utilise pas Home Assistant, sorry

pg68390, 2022-07-16:
@Patbob86
[image]

utilises tu ce module ?

GSzabados, 2022-07-16:
@smartmatic, right click on the side of the page, and then Translate to English if you use Google Chrome:

[image]

smartmatic, 2022-07-16:
> Bon bin ya plus qu'à tout jeter!!
> plusieurs semaines d'énergie et de temps perdus....
> A mon niveau, petit test avec modif du code pour envoyer une trame mqtt avec la date/heure du démarrage du D1Mini et de chaque prélèvement des données.. résultat: le mqtt fonctionne bien, je reçois la trame dans le docker.
> .. mais aucune réponse du compteur à mes interrogations..
> bye bye! Bon courage à tous!!
> tout est à jeter... aucun support ... projet à oublier immédiatement!
> Dommage :-(

Can you please show me where you have adjusted the code? Was your adjustment related to MQTT?

smartmatic, 2022-07-16:
> @smartmatic, right click on the side of the page, and then Translate to English if you use Google Chrome:
> 
> [image]

Thanks for the hint. I will try it with Google Translator :-)

Patbob86, 2022-07-16:
>

[7 more replies]

closed 2022-07-16 by Patbob86