psykokwak-com/everblu-meters-esp8266 #4: Reboot after reading the counter

Patbob86, 2022-06-24:
Thanks for these infos! I'll see what I do..

_Originally posted by @Patbob86 in https://github.com/psykokwak-com/everblu-meters-esp8266/issues/3#issuecomment-1163344235_

Patbob86, 2022-06-24:
Example: It finds counter on 433.844604, gives the data correct... and the it reboots..
Does anybody can explain this?

size_byte=124  l_radian_frame_size_byte=171
GDO0! (0, 0) TMO on REC
Test frequency : 433.844604
MARCSTATE : raw:0x13  0x13 free_byte:0x0F sts:0x02 sending 2s WUP...
202 free_byte:0x0E sts:0x07

size_byte=18  l_radian_frame_size_byte=25
GDO0! (0, 0) 1st synch received (1)  rssi=193 lqi=128 F_est=255 
GDO0! (1, 75) TMO on REC

size_byte=124  l_radian_frame_size_byte=171
GDO0! (0, 0) 1st synch received (1)  rssi=192 lqi=128 F_est=0 
GDO0! (1, 100) frame received (689)

------------------------------
Got frequency : 433.844604
------------------------------
Liters : 1051126
Battery (in months) : 240
Counter : 94

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Soft WDT reset

>>>stack>>>

ctx: cont
sp: 3ffffdb0 end: 3fffffc0 offset: 01a0
3fffff50:  00000017 43d8ec1c 3fff00ac 40202108  
3fffff60:  001009f6 0000005e 000000f0 00000006  
3fffff70:  00000012 feefeffe feefeffe feefeffe  
3fffff80:  0000005e 407b1d83 feefeffe feefeffe  
3fffff90:  feefeffe feefeffe feefeffe 3fff023c  
3fffffa0:  3fffdad0 00000000 3fff0228 40211ebc  
3fffffb0:  feefeffe feefeffe 3ffe8654 40100fa5  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,2)

load 0x4010f000, len 3460, room 16 
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4 
tail 4
chksum 0xc9
csum 0xc9
v0005af90

Patbob86, 2022-06-24:
So crazy!! How to explain also that at each reboot, it has found 4 times the counter with correct reading of the datas, but on 4 different frequencies going from 433.804 to 433.844 

------------------------------
Got frequency : 433.844604
------------------------------
Liters : 1051126
Battery (in months) : 240
Counter : 94
------------------------------
Got frequency : 433.828491
------------------------------
Liters : 1051126
Battery (in months) : 240
Counter : 95
------------------------------
Got frequency : 433.816406
------------------------------
Liters : 1051126
Battery (in months) : 240
Counter : 96
------------------------------
Got frequency : 433.804321
------------------------------
Liters : 1051126
Battery (in months) : 240
Counter : 97

xekil, 2022-06-24:
Hello,

It is normal that it restarts after finding a correct frequency and it is also normal that it finds different frequencies because the sweep is wide.

Fill in the freqeuenc 433.8446 in the file and you will see that it works correctly.

Patbob86, 2022-06-24:
Ok thanks!! I'll try this on next Monday.. Now, counter has gone for holyday !

pg68390, 2022-07-04:
@xekil Hello, je reviens vers vous car j'ai un souci de redémarrage après chaque test de fréquence, le module redémarre et reteste la première valeur soit le 433.76f , avez vous une solution pour moi ? cdlt Phil