psykokwak-com/everblu-meters-esp8266 #12: Compiling error

Sakoche31, 2023-04-10:
Bonjour,

Problème de compilation/
J'ai changé : ligne 31 char *jsonTemplate par const char *jsonTemplate

J'ai toujours un problème  après changement.
Merci pour l'aide

Sakoche31, 2023-04-10:
j'ai supprimé Echo_debug sur toutes les lignes du fichier cc1101.cpp, il doit y avoir un problème sur la déclaration : 
void echo_debug(T_BOOL l_flag, const char *fmt, ...); du fichier utils.h

Sakoche31, 2023-04-12:
Bonjour,

Voici les modifications que j'ai fait pour corriger le problème de compilation :  
Fichier Utils.h ligne 13 : void echo_debug(T_BOOL l_flag, **const** char *fmt, ...);
Fichier Utils.cpp ligne 62 : void echo_debug(T_BOOL l_flag, **const** char *fmt, ...)
Fichier everblu.ino ligne 31 : **const** char *jsonTemplate = 

Merci

fraserp, 2024-02-01:
Confirmed as working
(Confirmé comme fonctionnant)