psykokwak-com/everblu-meters-esp8266 #15: Had it working... now doesn't (Frequency De-calibrated?)

Lime-Parallelogram, 2023-06-19:
Hi there, 
I started testing this meter reading project on Thursday. I thought I'd start off with the Raspberry Pi version (https://github.com/neutrinus/everblu-meters) since it would be quicker to debug in case it didn't work. I got it to work reasonably well and I got at least a couple of reliable readings. On Friday, I started trying to get it it to work with the ESP32 device. This wasn't working at all so I plugged back into the Pi and I'm fairly sure I managed to get another good reading. After this, I resumed testing on the ESP device, including trying the frequency search. 

Since it was getting late and probably nearing the end of business hours, I thought I'd pause testing there. It is now Monday and I've tried multiple times with the ESP Board, 2 different CC1011 modules and now it won't even work with the Pi (just reading 0L).

Any ideas about why it may have stopped working like this - I was wondering if maybe the frequency sweep has upset the calibration on the CC1011 modules; is this possible?

Many thanks in advance for your suggestions.

hallard, 2023-07-11:
looking at the code, GDO2 is not used (pin of ESP `setup()` as input but never read) so my guess only GDO0 is used.

According to [everblu_meters.h](https://github.com/psykokwak-com/everblu-meters-esp8266/blob/main/everblu_meters.h#L57) it is connected to GPIO5, so you just need to connect GDO0 to a free GPIO and setup the IO pin here.

Wemos Mini D1 board as defined pin D1 as GPIO5 [see here](https://www.wemos.cc/en/latest/d1/d1_mini_lite.html) so it should works out of the box, except if you use another ESP board or didn't selected WeMos D1 Mini in Arduino IDE, that may have D1 not using GPIO5

This is why I always use GPIO number and not Arduino name that depends on board type, like that I know what I'm doing.

NickD9, 2023-07-12:
Yeah I got that wrong. GDO2 isn't used at all, but set up in code for the CC1101 to send serial data (which again isn't used), and the low to high transitions fool the ESP into thinking the CC1101 has received data, which it may have, or may not. It wasn't until I put a logic analyser on the lines that I could see what was going on. Eventually I found that whilst the data was transmitted to the everblu unit OK, and it was replying, the CC1101 wasn't getting the replies (? better receive on the everblu), so I moved the ESP/CC1101 units higher (upper floor of my house, which has ground floor, 1st floor and 2nd floor) and used a better DIY antenna pointed directly at the everblu through the glass of a window, then it all worked, and has been reliable ever since (2 weeks).