psykokwak-com/everblu-meters-esp8266 #23: Automatic calobration and frequency shift conpensation disabled.

Tazmaniac, 2024-04-01:
Rapidly looking at the code it seems that two very important features of the CC1101 chip are not used/disabled : 
- Frequency Synthesizer Calibration auto calibration
- Frequency shift compensation
witch are all available in FSK mode.
Auto-calibration should be done manually at each startup or automatically (but pll locking state should be monitored before transmitting/receiving).
At present, static calibration values are pushed into cal registers which are specific to a chip/crystal/vreg board combo.
Reworking the chip init state/calib would render the manual approximate static freq compensation stage of your project useless and would greatly improve the RF sensitivity/reliability and frame transmission/reception success.

psykokwak-com, 2024-04-16:
Could you propose a PR with that improvement ?

Tazmaniac, 2024-06-13:
I don't have access to an everblu meter.
Will use your codebase and do the proposed improvement if I get access to one.

genestealer, 2025-10-29:
@Tazmaniac I have implemented this in my fork: https://github.com/genestealer/everblu-meters-esp8266-improved/pull/14